j-be/AutoBim #3: [FR] LEVEL_CORNERS_USE_PROBE

Guilouz, 2021-02-27:
What do you think about this : 

Add a option to level the bed corners using a probe. Instead of using the nozzle to paper ajust the screws, it will probe the first corner (or the center) and save the z position. Then will go to the next corner to check if is whitin tolerance range. If is in tolerance, will go to next corner. **If is not, will leave the probe deployed waiting the user to raise the bed util the probe is trigered.** When all corners are within tolerance, will present a option to execute bed leveling.

This is how the option #define LEVEL_CORNERS_USE_PROBE works in Marlin.

Just a button to run this functionnality like when clicking on 'Level Corner' on printer's screen when this option is activated in Marlin.

j-be, 2021-02-27:
Thank you very much for pointing my attention to that. I honestly didn't think Marlin does it that differently, than the video link in `README` shows. So I never checked what it actually does. Interesting approach...

I'm not sure if this all is possible purely via GCode, especially the "detect if probe triggers" part. Do you have any insight on that? Could you maybe try it out? If not, that would mean you need to interact with the plugin more, as it cannot automatically move to next corner, as it doesn't know when. That would be a no-go for me - I want to be able to do everything on the printer without interacting with OctoPrint.

On the bright side: It would remove the need for `M117`, as one can visually see when the probe triggers, so that would improve support on those sh***y new Creality LCDs, which would be great.

I hope I find some to look into it this weekend. In any case (purely by GCode or not), I'd be happy to add it as alternative approach if I get it to work somehow, and, if it turns out to be more user-freindly, faster, or more accurate, even make it the default.

Guilouz, 2021-02-27:
Yes there is no GCode for that (like G35 and Tramming Assistant who is different), it's hard coded. 

If _LEVEL_CORNERS_USE_PROBE_ is disabled : When go in Level Corner on LCD screen, you can choose each corners to use paper to adjust them manually

If _LEVEL_CORNERS_USE_PROBE_ is enabled : When go in Level Corner on LCD screen, the process is running. It's very precise and very practical because you just have to raise the bed until the pin of the bltouch is retracted.

This is a part of the code in Marlin : 

`#if ENABLED(LEVEL_CORNERS_USE_PROBE)

  #define VALIDATE_POINT(X, Y, STR) static_assert(Probe::build_time::can_reach((X), (Y)), \
    "LEVEL_CORNERS_INSET_LFRB " STR " inset is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.")
  VALIDATE_POINT(lf.x, Y_CENTER, "left"); VALIDATE_POINT(X_CENTER, lf.y, "front");
  VALIDATE_POINT(rb.x, Y_CENTER, "right"); VALIDATE_POINT(X_CENTER, rb.y, "back");

  #ifndef PAGE_CONTAINS
    #define PAGE_CONTAINS(...) true
  #endif

  void _lcd_draw_probing() {
    if (!ui.should_draw()) return;

    TERN_(HAS_MARLINUI_U8GLIB, ui.set_font(FONT_MENU)); // Set up the font for extra info

    MenuItem_static::draw(0, GET_TEXT(MSG_PROBING_MESH), SS_INVERT); // "Probing Mesh" heading

    uint8_t cy = TERN(TFT_COLOR_UI, 3, LCD_HEIGHT - 1), y = LCD_ROW_Y(cy);

    // Display # of good points found vs total needed
    if (PAGE_CONTAINS(y - (MENU_FONT_HEIGHT), y)) {
      SETCURSOR(TERN(TFT_COLOR_UI, 2, 0), cy);
      lcd_put_u8str_P(GET_TEXT(MSG_LEVEL_CORNERS_GOOD_POINTS));
      IF_ENABLED(TFT_COLOR_UI, lcd_moveto(12, cy));
      lcd_put_u8str(GOOD_POINTS_TO_STR(good_points));
      lcd_put_wchar('/');
      lcd_put_u8str(GOOD_POINTS_TO_STR(nr_edge_points));
    }

    --cy;
    y -= MENU_FONT_HEIGHT;

    // Display the Last Z value
    if (PAGE_CONTAINS(y - (MENU_FONT_HEIGHT), y)) {
      SETCURSOR(TERN(TFT_COLOR_UI, 2, 0), cy);
      lcd_put_u8str_P(GET_TEXT(MSG_LEVEL_CORNERS_LAST_Z));
      IF_ENABLED(TFT_COLOR_UI, lcd_moveto(12, 2));
      lcd_put_u8str(LAST_Z_TO_STR(last_z));
    }
  }

  void _lcd_draw_raise() {
    if (!ui.should_draw()) return;
    MenuItem_confirm::select_screen(
      GET_TEXT(MSG_BUTTON_DONE), GET_TEXT(MSG_BUTTON_SKIP)
      , []{ corner_probing_done = true; wait_for_probe = false; }
      , []{ wait_for_probe = false; }
      , GET_TEXT(MSG_LEVEL_CORNERS_RAISE)
      , (const char*)nullptr, NUL_STR
    );
  }

  void _lcd_draw_level_prompt() {
    if (!ui.should_draw()) return;
    MenuItem_confirm::confirm_screen(
      []{ queue.inject_P(TERN(HAS_LEVELING, PSTR("G29N"), G28_STR));
          ui.return_to_status();
      }
      , []{ ui.goto_previous_screen_no_defer(); }
      , GET_TEXT(MSG_LEVEL_CORNERS_IN_RANGE)
      , (const char*)nullptr, PSTR("?")
[... 70 lines cut ...]
      }

      if (bed_corner != center_index) good_points++; // ignore center
      if (++bed_corner > 3) bed_corner = 0;

    } while (good_points < nr_edge_points); // loop until all points within tolerance

    ui.goto_screen(_lcd_draw_level_prompt); // prompt for bed leveling
    ui.set_selection(true);
  }

#else // !LEVEL_CORNERS_USE_PROBE

  static inline void _lcd_goto_next_corner() {
    line_to_z(LEVEL_CORNERS_Z_HOP);

    // Select next corner coordinates
    _lcd_level_bed_corners_get_next_position();

    line_to_current_position(manual_feedrate_mm_s.x);
    line_to_z(LEVEL_CORNERS_HEIGHT);
    if (++bed_corner >= available_points) bed_corner = 0;
  }

#endif // !LEVEL_CORNERS_USE_PROBE`

j-be, 2021-02-27:
Ok, so I all I can do is:

* Go to corner
* Measure
* If ok, goto next corner
* Else:
  * `M401`; Deploy the probe
  * `G0 Z0`; Move to Z=0
  * ... ???

Any suggestions on what `???` should look like? Is there anything (anything at all) in OctoPrint's Terminal when the probe retracts? I cannot currently check...

If not, dialog on OctoPrint maybe? Hardcoded timeout (i.e. just wait 5 seconds, how long can it take)?

Guilouz, 2021-02-27:
#if ENABLED(LEVEL_BED_CORNERS)
  #define LEVEL_CORNERS_INSET_LFRB { 40, 40, 40, 40 } // (mm) Left, Front, Right, Back insets
  #define LEVEL_CORNERS_HEIGHT      0.0   // (mm) Z height of nozzle at leveling points
  #define LEVEL_CORNERS_Z_HOP       4.0   // (mm) Z height of nozzle between leveling points
  //#define LEVEL_CENTER_TOO              // Move to the center after the last corner
  #define LEVEL_CORNERS_USE_PROBE
  #if ENABLED(LEVEL_CORNERS_USE_PROBE)
    #define LEVEL_CORNERS_PROBE_TOLERANCE 0.1
    #define LEVEL_CORNERS_VERIFY_RAISED   // After adjustment triggers the probe, re-probe to verify
    //#define LEVEL_CORNERS_AUDIO_FEEDBACK
  #endif

  /**
   * Corner Leveling Order
   *
   * Set 2 or 4 points. When 2 points are given, the 3rd is the center of the opposite edge.
   *
   *  LF  Left-Front    RF  Right-Front
   *  LB  Left-Back     RB  Right-Back
   *
   * Examples:
   *
   *      Default        {LF,RB,LB,RF}         {LF,RF}           {LB,LF}
   *  LB --------- RB   LB --------- RB    LB --------- RB   LB --------- RB
   *  |  4       3  |   | 3         2 |    |     <3>     |   | 1           |
   *  |             |   |             |    |             |   |          <3>|
   *  |  1       2  |   | 1         4 |    | 1         2 |   | 2           |
   *  LF --------- RF   LF --------- RF    LF --------- RF   LF --------- RF
   */
  #define LEVEL_CORNERS_LEVELING_ORDER { LF, RF, RB, LB }
#endif
  
In fact the process probe the 4 points first (or 5 if LEVEL_CENTER_TOO is enabled), it keeps the highest value of them to adjust others in tolerance defined by LEVEL_CORNERS_PROBE_TOLERANCE. If tolerance is good it goes to the next point if not it ask to user to move up the corner of the bed until the probe retracts or to skip this point, and it can also do another probe to see if tolerance is good when LEVEL_CORNERS_VERIFY_RAISED is enabled

The goal is to have all 4 corners within the defined tolerance.

There is no info in Terminal during the process

Here a video of the process : https://www.youtube.com/watch?v=FMbsn1hwxnU

j-be, 2021-02-28:
I'm not sure anymore I understand what you are getting at.

If you are just telling me the feature exists: Understood, I like the approach they are using. I already recommend to check the feature out as one of the first thing in `README` - I just never did, because it doesn't seem to work on Marlin `2.0.7.2` and I'd rather not switch to `bugfix`.

If you are just telling me to replicate the feature in my plugin: I would, but I don't think it is possible. In OctoPrint, I only can communicate with the printer using GCodes as far as I know. I can replicate everything up until [this point](https://www.youtube.com/watch?v=FMbsn1hwxnU&start=40) (i.e. second 40 of your video where the user needs to turn the knob). But in order to do so I need to use [`M401` - Deploy Probe](https://marlinfw.org/docs/gcode/M401.html) (seems to be the only way to deploy the probe without moving the printhead) - and if the probe is triggered after a `M401` it seems to go into error state. So I do not consider that to by an option. And even if it would be fine, I get no response on console, so I'm stuck there.

Of know Marlin can easily do it in code: it is nothing more than a `G30` without Z axis movement. If you know a way to replicate the behavior *using only GCodes* let me know, and I am happy to implement it.

On a different note: Please stop copy and passing hunks of codes in comments. Try to link it - you can even highlight code hunnks using link parameters like e.g. https://github.com/MarlinFirmware/Configurations/blob/62338c884318e91c4fc139402dfbb6ccc1ccfa46/config/default/Configuration.h#L1508-L1521 (notice the `L1508-1521` at the end of the link). That makes sure I can more easily read through it as I have syntax highlighting and all that sweet stuff. It also allows me to more easily check out the code in VSCode because I know where to find it. 

If you absolutely must copy-and-paste it, please make sure to properly escape it using backticks as described [here](https://support.zendesk.com/hc/en-us/articles/203691016-Formatting-text-with-Markdown-standard-agent-interface-#topic_xqx_mvc_43__row_fs1_zln_1n).

j-be, 2021-12-28:
Closing due to inactivity.

closed 2021-12-28 by j-be