j-be/AutoBim #4: UBL causes 3-point leveling at start of tramming

Fury1SOG, 2021-04-07:
Hello,
I recently installed Jyer's Marlin fork (https://github.com/Jyers/Marlin) on my Ender-3V2. The firmware supports UBL.

I noticed that when AutoBim starts tramming, a G29 J command is executed.
When G29 J is executed in an UBL-activated firmware, a 3-point leveling action is performed instead of turning off levelling compensation. (https://marlinfw.org/docs/gcode/G029-ubl.html)

As far as I can tell from the Marlin docs, a G29 P0 is the equivalent command for UBL.

j-be, 2021-04-07:
Ouch, interesting... Do you know if there is any way to determine whether UBL is enabled or not? And can you provide the output of `M115`?

j-be, 2021-04-07:
I implemented a setting to enable UBL support. Let me know if it works.

If you know of a way how to determine if UBL is enabled let me know. I'd be happy to add support for it.

Fury1SOG, 2021-04-08:
Thank you very much! I confirm that when the UBL checkbox is on, a G29 P0 is executed.

Here are the results of `M115`:

```
Send: M115
Recv: FIRMWARE_NAME:Marlin E3V2-Extensible-UI (Apr  6 2021 20:02:09) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V2 EXTRUDER_COUNT:1 UUID:xxxxxxxxxxxxxxxxx
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:1
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:1
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:1
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:1
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:1
Recv: ok
Send: M155 S2
Recv: ok
Send: M876 P1
Recv: ok
```

`M503` mentions Unified Bed Leveling in two locations:

```
Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E98.34
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X500.00 Y500.00 Z10.00 E60.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E5000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R1000.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X8.00 Y8.00 Z0.40 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Unified Bed Leveling:
Recv: echo:  M420 S1 Z10.00
Recv: 
Recv: Unified Bed Leveling System v1.01 active
Recv: 
Recv: echo:; Active Mesh Slot: 0
Recv: echo:; EEPROM can hold 1 meshes.
Recv: 
Recv: echo:; Servo Angles:
Recv: echo:  M281 P0 L10 U90
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H150.00 B45.00 F255
Recv: echo:  M145 S1 H190.00 B55.00 F255
Recv: echo:  M145 S2 H240.00 B70.00 F255
Recv: echo:  M145 S3 H240.00 B70.00 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P16.20 I1.08 D60.69
Recv: echo:  M304 P462.10 I85.47 D624.59
Recv: echo:; LCD Brightness:
Recv: echo:  M251 B255
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S1
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-43.50 Y-8.10 Z-2.30
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L0.00 U0.00
Recv: echo:; Filament runout sensor:
Recv: echo:  M412 S0 D0.00
Recv: ok
```

j-be, 2021-04-09:
Thanks a lot the console output. I threw together a quick routine to check it and automatically set `UBL` setting if not yet set. Would you feel comfortable testing that? The steps would be:

* Install the development version of this plugin (https://github.com/j-be/AutoBim/archive/next.zip)
* Log into your OctoPrint machine and from `~/.octoprint/config.yaml` delete the row `has_ubl: true`
* Restart OctoPrint
* Start the procedure
  * See what the message tells
* Abort the procedure
* Check in settings that `UBL` is checked

If you'd rather not - no worries, I'll throw together a UBL build eventually.

Fury1SOG, 2021-04-10:
Hello,
I am very happy to test! I performed the steps.
The "Seems like UBL system is active! If not, please change the setting." message was displayed and a G29 P0 is performed.

I noticed the following after the procedure:
1. `has_ubl: true` has been added back into `~/.octoprint/config.yaml`.
2. The UBL (Unified Bed Leveling) checkbox in the AutoBim plugin settings UI in the OctoPrint Settings is OFF.
3. A second procedure correctly performs G29 P0 even though the checkbox is OFF.

j-be, 2021-04-10:
@Fury1SOG thanks for the feedback, should be fixed now. Not sure if the plugin updates automatically while on `next` branch, so you may need to re-einstall (i.e. I think uninstall is not needed. Just reeinstalling using `Plugin Manager` -> `Get More...` -> `... from URL` should do the trick.

Fury1SOG, 2021-04-11:
I confirm that the UBL (Unified Bed Leveling) checkbox now turns on automatically. I will close this issue.
Thank you very much!

closed 2021-04-11 by Fury1SOG