j-be/AutoBim #8: Where is the button / screen?

HeMaN-NL, 2021-05-31:
I just installed this plugin in octopi 0.18.0 (with octoprint Version 1.6.1) on a Pi 3b+ connected to a Ender 3 v2 with Jyers firmware 1.32 with BLtouch and UBL.

In Octopi I did not see the plugin in the repository list, so I installed it by url as instructed in the readme.
I can see it installed; in the settings I can see AutoBim under the plugins. 
In the settings I enabled "show button in Navbar and restarted the Pi (all AutoBim sttings see below).

However, there is no button in the UI I can see, where is it supposed to be visible?
In the settings page - plugins AutoBim I can "Start AutoBim"and my printer start probing the four corners. The result of a probe is only visible in the "Printer Notifications"section of Octopi.

Is there anything I did miss or must change for the AutoBim button / UI to be visible?

```
AutoBim settings I have are:
invert arrows : off
Multipass: on
Accuracy 0.02
UBL: on
points: all default
show button in navbar: on
```

HeMaN-NL, 2021-05-31:
sorry, just found it; I did not notice the (blue) button at the top.
I also had to install some plugins to send messages to the display aparently, installed those and now I completed a first bed level with autobim

closed 2021-05-31 by HeMaN-NL

j-be, 2021-05-31:
Hey, that was faster then I was able to react :smile:

> I also had to install some plugins

I presume that is due to Creality doing LCD different from everybody else on the V2. Could you point me to the plugins? Maybe I can implement something so the plugins aren't needed.

HeMaN-NL, 2021-05-31:
https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

I think it was M73 progress plugin that did the trick, but I installed all other plugins mentioned as well