j-be/AutoBim #11: Does not complete check on Ender 3 Pro

KalixtShawxo, 2021-08-16:
Installed the plugin just fine.
ran it and levelled the first two points and then it just stops. says moving t the next point but does not.
Have restarted OctoPrint and the printer. retried and same happens.
Running Marlin 2.0.9.1 with a glass bed and BLtouch.

The Bed visualizer plugin works for my system, but thought this would be easier to make changes to the bed.

Thanks in advance.

j-be, 2021-08-16:
Please see [here](https://github.com/j-be/AutoBim#works-fine-on-the-first-or-first-and-second-corner-display-says-ok-moving-to-next-but-nothing-happens) - could this be the issue?

KalixtShawxo, 2021-08-16:
Hello Juri!
Thank you so much for addressing this.  I looked at the link, I have not changed the Ender 3 Pro bed sizing, I will check it after the current print job I have running completes.AutoBim is set to probe out only to 200 max.  and it does the first and second points.  so unless the X max is set too high for some reason, it should work right?  I have not had any problem with anything else going to the 200 point on the X. As i said, the OctoPrint Bed Visualizer plugin works fine, as does the BLTouch ABL. 
I will investigate this, and check the firmware as well.
Thanks again, Roger

KalixtShawxo, 2021-08-17:
Hello Yuri, The firmware settings are as below, direct screencap of the configuration.h files that I used to compile the firmware i am running. 

Is there something else I need to check?
Thanks again, Roger

ManuelS383, 2021-08-17:
I had the same Problem with my Ender5. I had to change the probing points to match my Probe position.
I am using an HeroMe5 FanDuct and after trying manuel the configured positions, by typing "G30 X30 Y200" into the Terminal, the Printhead doesn't move because this positions can't be reached with the duct. I've tried other positions and after changing the config to them it's running.

For me the following positions are working:
x 30 y 30
x 30 y 200
x160 y200
x160 y 30

KalixtShawxo, 2021-08-17:
Thanks. I guess I'm gonna have to try that next. Not seeing anything out of the ordinary in firmware.

KalixtShawxo, 2021-08-17:
I installed the new updated AutoBim that was available today. 
ok, I found my X maximum that will work with it is 191. so I'm resetting AutoBim to
30 3030 195190 195190 30
I got it to go around, I lost count of the number of cycles. Wish I could have gotten it right over the right side knobs but it seems to be working. 
One suggestion, the adjustments were hard to read on the printer screen, they scrolled from being so long  "-x.xx <<<< (Please adjust)"sometimes it was hard to see the - sign.  Might I suggest changing it to "-x.xx (Adjust up)" or "x.xx (Adjust down)"?  might make it easier to read the results. I was using the web browser to view the GCODE in the OctoPrint Terminal. 
Thanks for the help, now time to print something and test it out!
Roger

j-be, 2021-08-17:
@ManuelS383 Thanks for jumping in on this :smile: In the meantime I added some "Test" buttons to the corner points, so hopefully the manual `G30` is a thing of the past.

@KalixtShawxo I opened a dedicated issue for the scrolling text (see #12). Should be fixed shortly.

Thank you both for the feedback :smiley:

KalixtShawxo, 2021-08-17:
I'm back....I must have done something wrong. After AutoBim completed "done" and it was all level, I did the Bed Visualizer expecting to see an all green display and it was not.So I reran AutoBrim and i had made no adjustments after it was done... and it was now showing low just like the visualizer. 
I made all the recommended adjustments, got it "level" again per AutoBim. 

and then did the bed visualizer again and got this.  

So I'll admit I'm a newbie.....but one of these plugins is not telling the whole truth.
I had prior to using AutoBim did have an almost all green bed visualization except for one small bit of blue. Now I don't know where I stand cause the two plugins are definitely not telling the same story. 
:(  sorry to be a pain

KalixtShawxo, 2021-08-17:
Great!  I am glad I can help a little, although I'm still not getting what I thing I should be overall. see my last post I wrote before I saw your reply.

KalixtShawxo, 2021-08-17:
and the print job I just tried to do...total failure. the brim didn;t even stick and was mostly down and then just pulled upas one whole piece as the head mad a corner turn.

j-be, 2021-08-17:
I'm pretty sure by now you are trying to add some pictures/screenshots, but I don't see them here on GitHub.

Anyway, that is weird: both BedVisualizer (huge fan of that plugin myself) and AutoBim only display what Marlin tells them. So if both show something different, that means one of two things:

* Something in Marlin changes between the measurements, causing it to show different values when nothing changed
* Something in your printer changes between the measurements, causing different values

As the sooner will be extremely difficult to debug, I'd try to focus on the latter first:

Please try the following:

* Run BedVisualizer
* Screenshot the result
* Home the printer
* Run BedVisualizer again
* Compare to screenshot

This should give the exact same picture for each run. If not: you may have an issue with either Z axis or wobbly X gantry. See [here](https://www.youtube.com/watch?v=hgfTVNBRRTo) for the sooner and [here](https://www.youtube.com/watch?v=4bFYH0X3qjk) for the latter.

If it does: what board / firmware / probe are you rocking? Are you using "old" bilinear bed leveling, or the more modern UBL (aka. Universal Bed Leveling)?

KalixtShawxo, 2021-08-18:
ok, so the screen caps don't show up. gotcha.
anyway, I had an almost all green level with Bed Visualizer before using AutoBim. I have been working the last 4 hours just trying to get the bed level to where it was before I had made all the changes. Still chasing my tail for some reason.
I will do what you suggested and run the Bed Visualizer 2 times with no changes to see if it alters itself. 
Then I'm done for the night as I have to get ready for work.  
thanks

KalixtShawxo, 2021-08-18:
ok, being the overachiever I did three runs. each 3 minutes apart. and they all look identical to me.

KalixtShawxo, 2021-08-18:
If it does: what board / firmware / probe are you rocking? Are you using "old" bilinear bed leveling, or the more modern UBL (aka. Universal Bed Leveling)?

I have Ender 3 Pro with a 4.2.2. board, Marlin 2.0.9.1 firmware compiled to give me BLTOUCH and the higher z-axis kit (which I have not installed yet, just came today.)The about screen says bilinear leveling, but if I need to change something then I will.  I'm already compiling firmware cause ender does not produce factory firmware to support the higher z axis kit with BLTOUCH. I was hoping to get some printing done today, then install the z axis kit tomorrow along with longer cables kit. So pretty much I'll be rebuilding it tomorrow.

KalixtShawxo, 2021-08-19:
Well, I've been trying to relevel the bed for almost 2 days now. I pretty much disassembled the whole thing to put in the new higher z axis and wiring extensions, and still not able to get a good level. I keep running into the x0 y200 position being reported low, and I am not able to raise it any higher. the knob is completely loose. 
So I just upgraded the firmware to include UBL and am going to see if that makes any difference on anything. Have not gone back to AutoBim yet, still just trying to get Bed Visualizer to give me what I had before.

closed 2022-01-12 by KalixtShawxo