j-be/AutoBim #13: Starting z offset problem (maybe due to low center on bed)

lnevo, 2021-08-18:
Please remove all sections not relevant to this issue (including this statement).

## Bug report

Please do not report the "Works fine on the first (or first and second corner), display says 'ok. moving to next' but
nothing happens" bug. This is covered in detail in the "Troubleshooting" section of the README. I know it is an issue
for many, and it is on the roadmap, but unfortunately it isn't an easy fix.

### Description

I was using the Jyers firmware Probe Assisted tramming which has a thousandth mm precision and had the bed pretty level. When I tried using autobim it would tell me I had a +.08 offset at the first corner. After leveling everything and it doing a full round and saying done. I tried to do another round to make sure it was good and it started off saying it was .07 mm again. 

### Steps to reproduce

Checked with manual level to see where the corners were in relation. Ran the Autobim process through again. Verified manually leveling precision and starting autobim a 3rd time. Manual leveling was +.001/.000/.000/-.008. After starting autobim a 3rd time it reported again a .007 difference.

### Expected outcome

I would have expected it to run around all 4 corners and confirm that the autobim process was good. I would like to see a  probe of all 4 corners similar to the Probe Assisted Tramming and an offset target based on all 4 corners. I believe the fact that I have a low center is causing it to want to raise the bed each time.
 
### Actual outcome

Repeated use of autobim in my situation will cause the bed to continually be raised. 

### Environment (leave empty if unknown or not relevant)

* OctoPrint version: 1.6.1 
* Python version: 3.7.3
* Marlin version: Jyers 1.3.5b
* Probe (e.g. BL-Touch, EZABL, ...): BL Touch
* Printer (manufacturer/moodel, and please state any relevant modifications, like e.g. aftermarket mainboard): Ender 3 v2
* Browser (especially useful on UI issues): Chrome

For UI issues a screenshot always helps.

## Feature

**Please note by far the quickest way to get a new feature is to file a Pull Request.**

Just describe what you would like AutoBim as detailed as you can.

## Question

Make sure your question is not covered by [the Troubleshooting section](https://github.com/j-be/AutoBim#troubleshooting).

Else: just fire away. But keep in mind: the more detail, the easier it is to answer :smiley:.

j-be, 2021-08-24:
Hi, I think you got the issue right: I assume you use the probe for Z homing. If so, your bed's center is most probably some 0.07 to 0.08 mm higher (or lower) than the edges. The plugin tries to bring the corners up to level with the reference point (i.e. `Z0`, aka. Z home). While doing so, it obviously moves the center, too. When then homing again, the center will have shifted up by those `0.07` mm, causing the corners to be lower again.

The first important thing to note is, that your bed is still level at this point - it is just warped in the center.

But I presume this could be worked around, by allowing a point different from `Z0` as reference. I already though of this, as it may even speed up the process if one of the corners would be reference (i.e. that only leaves 3 to adjust). Do you think implementing that feature would solve your issue?

lnevo, 2021-08-24:
Precisely, we’re on the same page. I think a corner could be a reference, but which corner and how to tell if it’s high/low relative to the rest. That’s why I suggested a round of probes to get the tilt. A configurable reference could be an easy work around. I think you’d still need to adjust all 4 points, but maybe in practice it could work. I’d be happy to test.

j-be, 2021-08-24:
It is still possible though to use a Z endstop for Z homing - even if a probe is present. In those scenarios, `Z=0` is all I have, as I must not lower the bed below the endstop, no matter what. That is the reason I chose `Z=0` as reference: it works fine for both Z endstop and homing using probe.

But if (and only if) probe is used for Z homing, and using a corner as reference (the easiest would be to use the first corner - which could be set to any of them) I would rotate the bed (like: tramming is basically just rotating the bed in 3D) around a corner, which is part of the probing cycle, rather then one which is not - assuiming for most firmwares out there the Z homing reference will be "center of bed", even though it can be set to any point on the bed in Marlins configuration.

A third approach would be to make the reference point configurable. But for now I don't see any benefit in that. If one thinks that there is a special point on (or even off) the bed, which is especially good as `Z=0` sHe should set that as point for Z homing, too, right? And at that point my current implementation works as a charm again.

TL;DR: Using one of the corner as reference is on my TODO. It will most probably just be the first of the list.

lnevo, 2021-08-24:
Sounds like a plan. I’ll have to look more into setting up z stop with the BLtouch. So assuming then the endstop would get defined by the endstop instead of probe?

lnevo, 2021-08-31:
Just an FYI, I've added an "offset" variable in the settings and am adding that value in the try section. It seems to be working nicely. I can now level the bed, it will make it's rounds and when I execute the autobim process again, it will just cycle through all the corners as expected. This might be an easier way to go then changing the reference behavior and solves my problem for now. My real issue is the warped bed :)

j-be, 2021-12-28:
Closing due to inactivity.

closed 2021-12-28 by j-be