j-be/AutoBim #14: 3 Point leveling for Ender2

togeis, 2021-08-19:
Autobim requires a 4 point leveling? My Ender2 has a 3 point leveling heatbed. Is it possible to make the number of points configurable?

j-be, 2021-08-24:
Thanks for the suggestion. It is on my TODO list - didn't think it would come up that early. I will implement this as soon as I find a bit more time.

In the meantime I'd suggest setting 2 of the points to the same value. Not the nicest of workaround, but it should do the trick and not cause too much additional delay.

j-be, 2021-10-09:
@togeis Sorry for the delay, I have been on holiday in the meantime, and things came up, and so on...

In any case: I implemented two buttons in settings allowing to add and remove corner points. This should allow for an arbitrary number of points. It is included in release 0.1.11.

Would you mind checking it and (assuming the feature works for you as well) close this issue? Thanks :smiley:

j-be, 2021-12-28:
Closing the issue due to inactivity.

closed 2021-12-28 by j-be