j-be/AutoBim #15: Allow a timer inbetween probes

eblieb, 2021-08-20:
Would be nice to be able to make the pause in-between the probing of 1 corner longer. It probes twice and then really only shows the results when it is ready to probe again which gives you a very small window to put the corrections in.

lnevo, 2021-08-22:
I agree, it's not clear when the cycle starts and finishes all the time, plus it takes a few beats for the result to show on screen. I've had to jump away a few times when the test comes back within the threshold and it starts moving before I realize it was good.

j-be, 2021-08-24:
@eblieb @lnevo What kind of probe and printer are you using?

On my Ender 3 Pro with BL Touch 3.1 probing is way slow enough for adjustments in between 2 probes, i.e. I get some like 2 seconds between "I see the value on screen" and "probe touches bed for the next measurement".

Ukchris79, 2021-08-24:
having an option to delay or even wait for a user to push the button would be great, I do thing that sometimes the silicon spacers need time to settle between moves. I have increased the delay between moving to the next point, which has saved my fingers being pulled off.

Also you may want to think about the option to heat the bed with an adjustable time

eblieb, 2021-08-24:
Running an Ender 3 V2 and CRTouch probe and I dont get anywhere near 2 seconds inbetween value on screen and start next probe. Usually it is 1-2 seconds after the probe finishes that I see the value and by then it is already starting the next probe. 

Option to confirm before next probe would be very useful.

lnevo, 2021-08-24:
> @eblieb @lnevo What kind of probe and printer are you using?
> 
> On my Ender 3 Pro with BL Touch 3.1 probing is way slow enough for adjustments in between 2 probes, i.e. I get some like 2 seconds between "I see the value on screen" and "probe touches bed for the next measurement".

I think I have a 3.2 but not sure at all. Running jyers 1.3.5b i believe with 15x15 ubl. The first probe goes then the second, the results come on my screen as the probe is descending so it leaves just a little time to realize. I know thats subjective, I’m away from my printer till the weekend.

lnevo, 2021-08-24:
Yes, the sequence is what I’ve seen. No shift each probing.

eblieb, 2021-08-24:
I am going to hold off until Jyers releases a new firmware. M48 fails on 1.3.5 and 1.3.5.b (UBL both).

https://github.com/Jyers/Marlin/issues/1321#issuecomment-901716866

Looks like there is a STM32 issue in Marlin that was recently patched.

lnevo, 2021-08-31:
Found the setting to add a delay before moving to next corner. That took care of this issue for me.

tkunchick, 2023-06-24:
I am using an ender 3 with SKR3 without a screen, just octoprint as an interface, and I fully admit I am kinda slow. being able to adjust the time between probes on the same corner / location would be helpful. it seems it is already touching the bed and taking a reading again while I am trying to adjust from the previous reading.

As a bed tramming utility, and not something that needs to be ran before each print, a delay or slower bim, is not necessarily a bad thing.

closed 2023-06-24 by j-be

j-be, 2023-06-24:
I just released version 0.4.0, which contains the feature. OctoPrint should offer the update shortly (iirc. it caches plugin information for 24 hours).

Let me know what you think :smile:

tkunchick, 2023-06-25:
> I just released version 0.4.0, which contains the feature. OctoPrint should offer the update shortly (iirc. it caches plugin information for 24 hours).
> 
> Let me know what you think 😄

Tried it last night. Works awesomely. For me, seems 4 seconds gives me time to adjust and get out of the way before it probes again. 

Thank you very much.

j-be, 2023-06-25:
My pleasure

tkunchick, 2023-06-26:
OK, did some more testing. 

works great, But anyone using gantry leveling must make sure they are using the same X coordinates in both auto bim and g34.

I was going round and round watching terminal. g34 would move the gantry .07. then when I auto bim, it would ask me to move the screw .07 on the right side. 

my x axis was slightly different in g34, changeable via terminal with m422. I change them to be the same as what I was using for auto bim, and it's perfect now.

again, thank you very much for this great utility.

commit b46cf38f (2023-06-24): [Feature] Add "Delay between probes".

fixes #15