j-be/AutoBim #16: Something weird is going on

eblieb, 2021-08-24:
Recv: Bed X: 30.00 Y: 30.00 Z: 0.04
Recv: X:74.00 Y:37.00 Z:11.45 E:3158.73 Count X:5915 Y:2969 Z:4573
Recv: ok
Send: M117 0.02 >>> (adjust)
[...]
Recv: //action:notification 0.02 >>> (adjust)
Recv: ok
Send: G30 X30 Y30
[...]
Recv: Bed X: 30.00 Y: 30.00 Z: 0.03
Recv: X:74.00 Y:37.00 Z:11.45 E:3158.73 Count X:5915 Y:2969 Z:4573
Recv: ok
Send: M117 0.04 >>> (adjust)
Recv: //action:notification 0.04 >>> (adjust)
Recv: ok
Send: G30 X30 Y30
[...]
Recv: Bed X: 30.00 Y: 30.00 Z: 0.03
Recv: X:74.00 Y:37.00 Z:11.45 E:3158.73 Count X:5915 Y:2969 Z:4573
Recv: ok
Send: M117 0.03 >>> (adjust)

Been having issues where I can go around and around and it never stops having me adjust everything so pulled the terminal logs. Looks like there is something wrong in the plugin because it goes Z0.04 telling me to adjust 0.02 and then Z0.03 telling my to adjust 0.04... How is that possible?

eblieb, 2021-08-24:
Recv: Bed X: 30.00 Y: 30.00 Z: 0.03
Recv: X:74.00 Y:37.00 Z:11.45 E:3158.73 Count X:5915 Y:2969 Z:4573
Recv: ok
[...]
Send: M117 0.04 >>> (adjust)
Recv: //action:notification 0.04 >>> (adjust)
Recv: ok
Send: G30 X30 Y30
[...]
Recv: Bed X: 30.00 Y: 30.00 Z: 0.03
Recv: X:74.00 Y:37.00 Z:11.45 E:3158.73 Count X:5915 Y:2969 Z:4573
Recv: ok
Send: M117 0.03 >>> (adjust)
Recv: //action:notification 0.03 >>> (adjust)
Recv: ok
Send: G30 X30 Y30

There has to be sound rounding errors going on here.

j-be, 2021-08-24:
Your `M48` result looks as good as they get on a BL Touch.

I think this could be a "shifted by one", i.e. it always shows the value from the previous measurement, in your case:

```
Recv: Bed X: 30.00 Y: 30.00 Z: 0.04
Send: M117 0.02 >>> (adjust) // not sure where from
[ ... ]
Recv: Bed X: 30.00 Y: 30.00 Z: 0.03
Send: M117 0.04 >>> (adjust) // the 0.04 from above
[ ... ]
Recv: Bed X: 30.00 Y: 30.00 Z: 0.03
Send: M117 0.03 >>> (adjust) // this is actually the 0.03 from above
```

Could you maybe try to confirm that? I'll have a look at what's going on here. I thought I had enough countermeasures in place for that - seems like I don't.

eblieb, 2021-08-24:
Currently in a print but will have it go to 1 spot and then randomly change the level and see what it reports.

eblieb, 2021-08-24:
One thing I noticed... You use G29 D to turn off the ABL but that only works in UBL... If running Bilinear the D flag has it run a dry run on the ABL.  Should use M420 S0 to turn off mesh

eblieb, 2021-08-24:
I switch to a ABL firmware and unchecked the "use UBL" in the settings and it is working great now. So the issue exists with UBL firmware and the UBL check box enabled.

j-be, 2021-08-24:
Interesting... will see what I can find out. Still think this may be caused by the same issue that causes #15.