j-be/AutoBim #17: Logic Issue

eblieb, 2021-08-24:
So there is a logic issue with the testing methodology implemented. 

The issue is when setting the threshold is set at the lowest resolution that the firmware is reporting. 

I tested my probe and got
Recv: Mean: 0.008478 Min: 0.007 Max: 0.009 Range: 0.003
Recv: Standard Deviation: 0.000844

but usually the range is closer to 0.006

But because the firmware only outputs 2 decimals, there is an inherent rounding issue where that range is really 0.02 because it can be + or - and is statistically random when it can occur. 

A recommendation to help this issue (wont totally mitigate it) would be to have the AutoBIM test the point 2 separate times and compare the z value, if the z value matches than that can be reported for the adjustment. If they don't match then probe a 3rd time and average the values and round. That way if 1 comes back 0.01 and the other 0.00 if the 3rd comes back 0.01 then it would round to 0.01 but if the 3rd comes back as 0.00 it would round to 0.00. This will help prevent the issue of constantly going around and around the bed because of this possible rounding issue and having the autobim never finish unless manually cancelled.

j-be, 2021-08-24:
That is what the `Accuracy threshold` is for - and it has a bit more layer than that.

> But because the firmware only outputs 2 decimals

True for most configurations, but actually there is a possibility to change that in Marlin (see [here](https://github.com/MarlinFirmware/Configurations/blob/6f1ce9abf07d89e5ed09d7cea818c8cc1bbd7bea/config/default/Configuration_adv.h#L2247)) - AutoBim should work fine with higher resolutions.

> but usually the range is closer to 0.006

True, but `-0.0049` to `0.0049` (which will both round to `0`) is still a range of `0.098`. So a threshold of `0.01` should still work fine here. See #1 for my thoughts on this.

> having the autobim never finish unless manually cancelled.

I've used the plugin since February, and while I sometimes needed quite some rounds (from memory I'd say worst case up to 10 rounds, or maybe even more, normal case something like 4 to 5) it always came to an end eventually. But yeah, as long as you only see 2 places behind the decimal `0.005` and `0.014` will both be displayed as `0.01`. And adjusting one corner will inevitably change all others, except maybe if you are exactly (and I mean exactly-exactly) above the center of each leveling screw. So going for a `0.01` threshold on such a setup has its challenges. I still manage, but I think I will higher it to 3 places next time I rebuild and reflash Marlin.

j-be, 2021-12-28:
Closing this issue due to inactivity.

closed 2021-12-28 by j-be