j-be/AutoBim #18: Add option to choose corner travel order

lordekeen, 2021-08-31:
First of all, thanks for the plugin, its awesome and really made my life easier since i don't have enough progmem to add anything else on Marlin.
I would like to suggest an option to choose the path between the corners, instead of only the "square" shape.
Something like this, lets assume the default order the head will move and probe is this one:
B------C           
|ㅤㅤㅤㅤ---x---|
A------D
From A to B, C and D.
The user could then input the path he/she wants to override, like an X pattern:  ACDB or ADBC.

j-be, 2021-08-31:
Hi, happy you like it 😃 

If you want it permanently: you can just swap 2 corners in AutoBim's settings.

Else: I could imagine having something like a "Choose pattern" setting - should be fun to implement, too. The possible patterns are only ABCD and ACBD, or am I missing something?

lordekeen, 2021-08-31:
Sorry i didnt notice that i could swap corners already.
As for the patterns, i can visualize three, ABCD (square), ACBD (X) and ADBC (Z), changing the first corner would kind of rotate the pattern (like BCDA etc.).

j-be, 2021-09-01:
No worries 😄 

You're right, `Z` is a valid unique pattern, too.

I'll add this to the roadmap, as I said: could be fun to implement.

lordekeen, 2021-10-10:
@j-be I've changed the order manually by editing the corners in the settings, maybe this functionality isnt really needed after all.