j-be/AutoBim #23: repeated runs don't measure the same

unitrix0, 2021-11-06:
I find it confusing that when I complete a run and immediately start another one it already wants to change the level again. If I've successfully completed a run and start a new run shouldn't it measure the same as before?

j-be, 2021-11-06:
Hi, could be same issue as reported in #13, i.e. warped bed. If so, try activating "First corner is reference (experimental)" in AutoBim's settings.

If not, please try to give a bit more context, like printer, board, firmware, probe, is probe Z endstop, `M48` (https://marlinfw.org/docs/gcode/M048.html), AutoBim Settings, stuff like that. Without any details it is really hard for me to figure out what goes wrong here.

unitrix0, 2021-11-18:
I tried the "First corner is reference" setting and this actually seems to help. It also takes much less time to level the bed. Actually, how I do it is, I keep restarting the measurment until the reference measurement differs  0.00 from the center point. Maybe this is not neccessary?

I already suspected the bed is bent. A new one is already ordered...

jus to provide this information too:
**Printer:** Ender 5 plus with BL Touch
*Board:** "Silentboard" with **firmware:** 1.70.2 BL

j-be, 2021-11-20:
> I tried the "First corner is reference" setting and this actually seems to help

Good to hear!

> until the reference measurement differs 0.00 from the center point

Well, if the bed is actually bent, it is imho. better to accept a difference between center and corner, while having all corners at the same height. Also, keep in mind that the bed will deform as it heats up, but the screws will hold those 4 points relatively stable. So you're chasing a moving target. It is imho. much more important, that the printer is square, and that includes a trammed bed.