j-be/AutoBim #24: Cannot probe over adjustment knobs on ender 3 pro

SpiderSpartan, 2021-11-18:
### Description

I cannot get AutoBim to probe the points over any of the adjustment screws via running or the test buttons. The probe position limits that do work seem arbitrary. It will not probe anything below X88 Y50 or above X235 Y197. The 235 I could maybe understand, since that is my bed size. I'm able to position the probe over the points I want using G0 without issue (ie: G0 X71 Y43 works just fine). Below are the firmware lines I could think might be relevant to this, but I feel like I'm just missing something. Is there another setting I should be looking at in the firmware or do the above limits make sense to anyone based on the below config info?

```
#define AUTO_BED_LEVELING_BILINEAR

X_BED_SIZE 235
Y_BED_SIZE 235

X_MIN_POS 0
Y_MIN_POS 0
X_MAX_POS (X_BED_SIZE + 5)                               = 240
Y_MAX_POS (Y_BED_SIZE + 5)                               = 240

Z_SAFE_HOMING_X_POINT ((X_BED_SIZE)/2) + 44	         = 161.5
Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE)/2) + 6	         = 123.5

NOZZLE_TO_PROBE_OFFSET {-44, -6, 0}

PROBING_MARGIN 10
```

### Environment (leave empty if unknown or not relevant)

* OctoPrint version: Version 1.7.2 
* Octopi:Version 0.18.0, running on Raspberry Pi 2 Model B Rev 1.1
* Python version:
* Marlin version: Marlin-bugfix-2.0.x
* Probe: BL-Touch 3.1
* Printer: Ender 3 Pro 
* Browser: Chrome

Sensor position issues aside, AWSOME work on this plugin. I never felt like I was doing quite good enough with paper or feeler gauges, even when doing multiple rounds, but with this plugin my first layers are looking awesome and sticking great. THANK YOU!

j-be, 2021-11-18:
Hi, thanks for the kind word, happy you like it.

My first guess is that it has something to do with probe-to-nozzle offset (see https://marlinfw.org/docs/gcode/M851.html, please add the result here if the following does not solve your issue): Marlin already factors those in when using `G30` (which is used by AutoBim). So the points in AutoBim's settings actually refer to where the probe is while probing, not where the nozzle is.

In other words:

* Home the printer
* Put a piece of paper at the center of the bed (painter's tape or a Post-It work even better, as they tend to move less)
* Move the nozzle there using `G0 X117 Y117`
* Mark the point exactly below the nozzle on the paper
* Use `G30 X117 Y117` (notice the same coordinates as above)

Now, if the offsets are right the BLTouch should touch down exactly at the marked point.

SpiderSpartan, 2021-11-18:
Thanks for the tip. Funny thing is, M851 gave me a Probe Offset X10.00 Y10.00 Z-2.87 despite configuring the firmware as NOZZLE_TO_PROBE_OFFSET {-44, -6, 0}. I did have Z probe offset wizard enabled, which is where the Z-2.87 comes from. I set the offset using M851 X-44 Y-7 and now G30 X117 Y117 probes the same point as G0 X117 Y117 puts the nozzle. Only problem now is, the probe limits didn't change, so the front right probe is well inside the adjustment screw. Based on my Min/Max positions and PROBING_MARGIN I'd expect my min probing position to be X54 (44+10) and Y 17 (7+10). Guess I need to dig into my firmware to determine why my probing limits aren't what I expect.

SpiderSpartan, 2021-11-18:
FOUND IT! A monster of my own making. Thanks for the help getting here.

```
#if PROBE_SELECTED && !IS_KINEMATIC
  #define PROBING_MARGIN_LEFT (88)  //(30 + 44)  //PROBING_MARGIN
  #define PROBING_MARGIN_RIGHT (0) //(30 - 44)  //PROBING_MARGIN
  #define PROBING_MARGIN_FRONT (50) //(30 + 6)  //PROBING_MARGIN
  #define PROBING_MARGIN_BACK (38)  //(30 - 6)  //PROBING_MARGIN
#endif
```

closed 2021-11-19 by SpiderSpartan

j-be, 2021-11-20:
Hi, haooy to help! Happy printing.