j-be/AutoBim #25: G30 coordinates ignoring probe offset

charredchar, 2021-12-13:
I am running Klipper, BLTouch is set up correctly with the X and Y offsets, following the Klipper documentation and `Bed_Mesh` works as expected. Added the G29 and G30 Macros to my Klipper config, both seem to respond to commands without an error.

When running through the Preparation section I used `G0 X150 Y150` for the middle of the bed. When issuing the `G30 X150 Y150` it does not move the probe over to the location of the nozzle, instead it probes in the same spot. If I was to issue `G30 X183 Y139` to take into account my probe offset it moves to where I expect.

* OctoPrint version: 1.7.2
* Python version: 3.7.3
* Klipper version: 0.9.1-490-g8eef1e21
* Probe (e.g. BL-Touch, EZABL, ...): BLTouch
* Printer (manufacturer/moodel, and please state any relevant modifications, like e.g. aftermarket mainboard): Creality CR-10
* Browser (especially useful on UI issues): Chrome/96.0.4664.93

datapaganism, 2022-01-04:
Can we get add to an option the settings for users to key in their own offsets and subtract the offset when calculating the difference?

datapaganism, 2022-01-09:
fixed as of commit https://github.com/j-be/AutoBim/commit/8ae2b5d25105c85ed636359a347e16844e0218b8,
@j-be  please close this issue thanks.

j-be, 2022-01-10:
@datapaganism Thanks, did not think of this.

For reference: This is was fixed with #29.

closed 2022-01-10 by j-be