j-be/AutoBim #26: Ender 3 Max - Point X30 Y300 seems to be unreachable!

MrSignalPlus, 2021-12-27:
I am running a E3 Max with LCD screen, CR touch and Jyers firmware, but cant get the plugin to work. When I press the tram button the printer homes and just hovers or tells me Point X30 Y300 seems to be unreachable!.

The extension works great on my 3V2, so I'm not sure what might be wrong, considering the motherboard, screen and firmware are the same.

j-be, 2021-12-28:
Hi, have you checked the point is actually reachable by the probe?

In other words: if you home all axis  and then enter a `G30 X30 Y300` in OctoPrint's Terminal: does the printhead move? If not: the point is most probably actually not reachable due to a combination of `X_MAX_POS`, `PROBE_MARGIN`, probe to nozzle offset, and others.

MrSignalPlus, 2022-01-01:
So I entered G30 X30 Y300 into my octoprint terminal and nothing happened

j-be, 2022-01-02:
So X30 Y300 is not reachable by the probe.

Be sure to check your probe-to-nozzle offset is correct (see https://marlinfw.org/docs/gcode/M851.html) and if everything is correct: your printer just can't move there to probe the point.

j-be, 2022-01-04:
As I was just discussing at #28 I have the wrong instructions for software endstops in the README: they are enabled/disabled with `M211`, not `M121` (see https://marlinfw.org/docs/gcode/M211.html).

The theory behind that: my Ender 3 Pro has a bed size of 235x235, but it can easily move to X=245 and Y=240 without colliding with anything.

Now, you can (and if you compile your own firmware you actually should) find those max values and set them as `X_MAX_POS` and `Y_MAX_POS` in `Configuration.h`, as it also e.g. increases bed levelling coverage in `G29`.

If not: you may experiment with `M211`, but be VERY CAREFUL while software endstops are disabled.