j-be/AutoBim #27: 'ok. moving to next' on first probe point when disabling Mulitpass probing

UnavengedAvo, 2021-12-27:
### Description

I know that this error had been reported before, but the process does run fine when _Mulitpass probing_ is **enabled**. I had gone through the steps in the [FAQ](https://github.com/j-be/AutoBim#works-fine-on-the-first-or-first-and-second-corner-display-says-ok-moving-to-next-but-nothing-happens), but they do not seem to fix the issue.

I seem to only see this issue when _Mulitpass probing_ is **disabled**. The probe gets the reference either from Z offset, or from the First Corner, it then moves on to the first probe point and then just gets stuck on that point. Looking at the terminal output, it seems as though the plugin is just repeating the same _G30 X28 Y198_ command:

```
Send: M117 wait...
Recv: ok

Send: G91
Recv: ok

Send: G28 X0 Y0 Z0
Recv: X:165.00 Y:129.00 Z:11.30 E:0.00 Count X:13200 Y:10320 Z:4520
Recv: ok

Send: G90
Recv: ok

Send: G0 Z20
Recv: ok

Send: G29 J
Recv: ok

Send: G30 X28 Y198
Recv: Bed X: 28.00 Y: 198.00 Z: 0.23
Recv: X:71.00 Y:207.00 Z:11.30 E:0.00 Count X:5680 Y:16560 Z:4520
Recv: ok

Send: M117 ok. moving to next
Recv: ok

Send: G30 X28 Y198
Recv: Bed X: 28.00 Y: 198.00 Z: 0.24
Recv: X:71.00 Y:207.00 Z:11.30 E:0.00 Count X:5680 Y:16560 Z:4520
Recv: ok

Send: G30 X28 Y198
Recv: Bed X: 28.00 Y: 198.00 Z: 0.25
Recv: X:71.00 Y:207.00 Z:11.30 E:0.00 Count X:5680 Y:16560 Z:4520
Recv: ok

Send: M117 Aborted by user
Recv: ok
```

### Steps to reproduce

I just disabled _Mulitpass probing_, and also tested both with _First corner is reference_ enabled and disabled.

Not sure whether it is related, but I had changed the order in which the points are probed as follows:
┌─────┐
│1............2│
│.................│ 
│4............3│
└─────┘
I confirmed beforehand whether these points could in fact be probed by following the steps in the [FAQ](https://github.com/j-be/AutoBim#works-fine-on-the-first-or-first-and-second-corner-display-says-ok-moving-to-next-but-nothing-happens).

The only way I found to fix this is by going to the _Plugin Manager_, and then clicking _Cleanup Plugin Data_.

### Expected outcome

I expected that disabling the _Mulitpass probing_ and enabling _First corner is reference_ would allow the plugin to check the reference point once and then probe the other 3 corners so that I can level them to that point. (Similarly to how G35 works)

### Environment

* OctoPrint version: 1.7.2 (OctoPi Version 0.18.0, running on Raspberry Pi 3 Model B Rev 1.2)
* Python version: 3.7.3
* Marlin version: 2.0.9.2
* Probe: BLTouch Smart V3.1 2019 Edition
* Printer: Creality Ender 3 Pro 2019 Edition
* Browser: Google Chrome Version 96.0.4664.110

j-be, 2021-12-28:
Hi, thanks for the feedback.

I need to dig a bit deeper to see what causes the issue you reported - not sure what could cause this.

In any case, the "Expected outcome" is not what the plugin is trying to do here, i.e. it is not a replacement for `G35`: In _Mulitpass probing_ mode the plugin will go round and round, until it does one full round without any need for adjusting. Without _Mulitpass probing_ it should just do one round - still each point interactively - and then stop.

_Mulitpass probing_ is recommended, as on most beds changing one corner will inevitably affect all others, too.

Also, I'm missing some `M117` in the log, i.e. the ones telling `0.02 >>>> (adjust)`. Did you clear those from the log, or weren't they there to begin with?

j-be, 2021-12-28:
Update: I found the issue, will release a bugfix shortly. It also explains why there aren't no `M117` in your log.

j-be, 2021-12-28:
Issue should be fixed in version `0.2.3`. Please update and try again and let me know whether it worked or not.

closed 2021-12-28 by j-be

UnavengedAvo, 2021-12-28:
At first I updated from the [Master](https://github.com/j-be/AutoBim/archive/refs/heads/master.zip) branch, and it stated that it was still on v0.2.2. I then swapped to the [Next](https://github.com/j-be/AutoBim/archive/refs/heads/next.zip) branch, and I am now on v0.2.3dev0.

I can confirm that it now completes one round of probing when _Mulitpass probing_ is disabled in the v0.2.3dev0 branch.

j-be, 2021-12-28:
My fault, got mixed up: above should read `0.2.2` - that is today's release. Sorry for that.

commit 98ba065e (2021-12-28): [Bugfix] Fix singlepass.

The check for multipass caused all messages to be swallowed. Should be
fixed now.

fixes #27