j-be/AutoBim #28: Ender 3 S1

paul-aviles, 2022-01-03:
Worked fine on an Ender 3 S1 stock firmware 1.0.3C but the bed measurements need to be adjusted to:

x 10 - y 25
x 20 - y 179
x 179 - y 179
x 179 - y 25

I think the probe points (hence the x10 - y25) should sit on top of the screws for better measurement/alignment but that is my personal feeling.

Juri, this is an amazing plugin so thank you!

j-be, 2022-01-03:
Hi, thank you very much.

You sure these coordinates are correct? Marlin *should* factor in probe-to-nozzle offset when doing a `G30`, so the coordinates actually should be where the screws are on the bed, not where the nozzle is while the printer is probing. The coordinates seem a bit odd for an Ender 3 sized machine.

paul-aviles, 2022-01-03:
Yeah, it is the only way to make the screws. See in the picture. 
[image]

Those are the max coordinates the unit can reach, otherwise the nozzle does not move. There was no change when using M121 to overwrite the points either.

j-be, 2022-01-04:
Ouch, seems like I got it wrong in the README: should be `M211 S0` to disable and `M211 S1` to reenable (see https://marlinfw.org/docs/gcode/M211.html). `M121` seem to be the physical endstops, not the virtual ones.

But be VERY CAREFUL when doing that!

There are 2 settings in Marlin, bed size and max coordinate. My Ender 3 Pro has a bed size of 235, but can easily move to X=245 without colliding with anything. If the same is true for the S1, using `M211` may help.

I'd be happy to implement support for it if you think that would be benificial.

paul-aviles, 2022-01-04:
Hmm, weird, see the results..

M211 S0; Disable End Stop
Recv: echo:Soft endstops: OFF  Min:  X-10.00 Y-8.00 Z0.00   Max:  X230.00 Y220.00 Z270.00
Recv: ok

M211 S1; Enable End Stop 
Recv: echo:Soft endstops: ON  Min:  X-10.00 Y-8.00 Z0.00   Max:  X230.00 Y220.00 Z270.00
Recv: ok

so I guess they should work, but the same behavior, only moves until Y179 not passed that. 

Send: G30 X30 Y190
Recv: ok
[...]
Send: G30 X30 Y180
Recv: ok
[...]
Send: G30 X30 Y179
[...]
Recv: Bed X: 30.00 Y: 179.00 Z: 0.00
Recv: X:61.80 Y:219.50 Z:13.25 E:3390.83 Count X:4944 Y:17560 Z:5300
Recv: ok
[...]

Here are the results of 503 just in case.

Send: M503
Recv: echo:  G21    ; Units in mm (mm)
Recv: 
Recv: echo:; Filament settings: Disabled
Recv: echo:  M200 S0 D1.75
Recv: echo:; Steps per unit:
Recv: echo: M92 X80.00 Y80.00 Z400.00 E424.90
Recv: echo:; Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:; Maximum Acceleration (units/s2):
Recv: echo:  M201 X500.00 Y500.00 Z100.00 E1000.00
Recv: echo:; Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P500.00 R800.00 T500.00
Recv: echo:; Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 B20000.00 S0.00 T0.00 X5.00 Y5.00 Z0.30 E5.00
Recv: echo:; Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:; Auto Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: echo:  G29 W I0 J0 Z0.33000
Recv: echo:  G29 W I1 J0 Z0.24750
Recv: echo:  G29 W I2 J0 Z0.19250
Recv: echo:  G29 W I3 J0 Z0.15125
Recv: echo:  G29 W I0 J1 Z0.14000
Recv: echo:  G29 W I1 J1 Z0.10250
Recv: echo:  G29 W I2 J1 Z0.09250
Recv: echo:  G29 W I3 J1 Z0.04000
Recv: echo:  G29 W I0 J2 Z0.07000
Recv: echo:  G29 W I1 J2 Z0.07750
Recv: echo:  G29 W I2 J2 Z0.10500
Recv: echo:  G29 W I3 J2 Z0.10000
Recv: echo:  G29 W I0 J3 Z0.10500
Recv: echo:  G29 W I1 J3 Z0.08000
Recv: echo:  G29 W I2 J3 Z0.05000
Recv: echo:  G29 W I3 J3 Z-0.00500
Recv: echo:; Material heatup parameters:
Recv: echo:  M145 S0 H215.00 B50.00 F255
Recv: echo:  M145 S1 H240.00 B100.00 F255
Recv: echo:; PID settings:
Recv: echo:  M301 P18.76 I1.34 D65.79
Recv: echo:  M304 P107.00 I20.86 D365.95
Recv: echo:; Power-Loss Recovery:
Recv: echo:  M413 S1
Recv: echo:; Z-Probe Offset (mm):
Recv: echo:  M851 X-31.80 Y-40.50 Z-3.25
Recv: echo:; Filament load/unload lengths:
Recv: echo:  M603 L0.00 U100.00
Recv: ok

paul-aviles, 2022-01-13:
Hey Juri, quick question on something I am noticing. Say I use AutoBim to set the Ender 3V2, it goes through the cycles and finally done. If I immediately do another one, the system things is out of calibration.

I just did for instance, and all was ok. I did the process again and the first corner was -0.11 out, that is a lot.

I have the regular 3nder 3V2 with the second stepper motor and the Y cable connecting both Z motors and for what I get the unit is pretty balanced and prints are pretty good, but this seems to be an issue with my unit.

Any ideas or suggestions or opinion on something like? https://www.youtube.com/watch?v=oUdoJUs8Zcg

Regards,

Paul Aviles

j-be, 2022-01-14:
If I had to guess I'd go for "bent bed" (see #10), but 0.11 is indeed a lot. Apart from using "First corner as reference" as propsed there too I'd recommend counterchecking with the [Bed Level Visualizer](https://plugins.octoprint.org/plugins/bedlevelvisualizer/).

As for the belt driven Z axis: I am most definitely going to do that next time I'm heavily bored. I'm just not sure if I'll go with the design you linked, or if I'll mod the Ender 3 to a Voron Switchwire (see [here](https://github.com/walttriano/Ender_3Pro_Switchwire) or [here](https://github.com/boubounokefalos/Ender_SW) or Google - there are a couple more).

I think though it is not necessary to get good prints, but it might help. Also, Z axis with screw is unbelievably difficult to troubleshoot imho. And it sure looks amazing.