j-be/AutoBim #31: Feature Requests: Bed Heating and Auto Stop

mlee12382, 2022-02-19:
## Feature
There are a couple of features that I would love to see added to AutoBim:
When starting the autobim routine it should heat up the build plate and stabilize the temp before probing for more accurate results, you could add settings to the main settings page to adjust the pre-heat temp and wait time once temp is reached before probing starts. This seems like it should be relatively easy to add.

The other feature that would be awesome to have though slightly harder to add due to the logic needed to make it work would be a routine that checks to see if it's made a complete 4 point set of probes without calling for any adjustments then it ends the autobim routine automatically. You would need to have some if / then checks and if any 1 point needs adjustment it resets the check and starts it over once it goes back to location 1. I'm sure with enough effort this should be possible to map out but my programming skills are pretty rusty and my octoprint plugin programming skills are non-existant lol. 

Thanks for creating an awesome plugin, I just wish I had found it sooner lol. I will definitely be sharing it with all of my friends.

j-be, 2022-02-19:
Hey, thanks for the kind words :smile: 

About the pre-heat: It wouldn't be as easy to implement this in a clean way as might one think. Maybe I am missing something, but from what I know by now there isn't an easy way in OctoPrint's plugin API to simply wait for something to occur.

Also: I doubt it would actually make a difference. I can see why one would want to heat the bed before invoking the `G29`, as while the bed heats up - due to it being fairly low quality aluminium on most printers - it may bulge. But I don't think it would change the geometric plane the bed is based on (not sure how to explain it).

Could you please check on your side if it actually makes a difference? It doesn't seem to make one for me. The way I check:

* Enable `First corner is reference` in settings. Only do this if you are using the Z probe as Z endstop. Do NOT use this setting - ever! - if you use a switch as Z endstop. That is btw. the reason why it is not on by default.
* Run AutoBim and trim the bed
* Heat the bed
* Run AutoBim again

As for the "abort on X consecutive points without adjustment" - yeah, I already thought of this. It wouldn't actually be that hard to implement - far easier than the "heat bed and wait", but I just never came around doing it. I'm sure I'll implement it eventually, just not sure when :smiley:

j-be, 2022-04-13:
@mlee12382 I just released `v0.2.3` with support for both your features.

Using the custom GCode for heating the bed will most probably cause a timeout in the current implementation, which is why I'd keep this issue open for now. I need to figure out how I can avoid that. I have an idea - we'll see if that works.

The AutoStop as you called it - i.e. stop after all points were measured correctly - should work now.

mlee12382, 2022-04-13:
Awesome, I will test it out, I've manually preheated the bed before running autobim and it didn't seem to timeout on me, it probably sees it as being active and not idle so the timeout isn't being triggered, will let you know if I have any issues.

j-be, 2022-04-13:
Oh, sorry, I meant AutoBim will most likely timeout as it doesn't take into consideration such a long delay. So keep an eye on OctoPrint - it should display an error as it times out, hopefully 😄