j-be/AutoBim #32: Inconsistent readings

jrburris, 2022-03-29:
### Description

Inconsistent readings
 
### Steps to reproduce

This happens often but not every time and not always on the same corner. When I adjust a corner, i'll get a message that says "0.05 >>>". If I don't touch anything and let it run a couple cycles, it will randomly jump to something like "0.09 >>>". I checked the frame and rollers, and everything is tight.

### Expected outcome

I would expect if I don't adjust anything, the reading would stay the same.

### Actual outcome

See description

### Environment (leave empty if unknown or not relevant)

* OctoPrint version: 1.7.3 
* Python version: 3.8
* Marlin version: IDK
* Probe (e.g. BL-Touch, EZABL, ...): BL-Touch
* Printer (manufacturer/moodel, and please state any relevant modifications, like e.g. aftermarket mainboard): Ender 3 V2 stock board
* Browser (especially useful on UI issues): Chrome

j-be, 2022-03-30:
Hy, that is very weird indeed.

Can you try to do an `M48` (https://marlinfw.org/docs/gcode/M048.html) and post the result here? I hope Creality has it activated in their firmware.

Additionally, please provide a log from the OctoPrint Terminal while AutoBim is running?

It may be the same issue as reported in #20.

jrburris, 2022-03-31:
Thanks for the quick response. Here is the log from Octopi. See that it jumped from "0.15 >>>” to "0.20 >>>” without me touching anything.

Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Send: M48
Recv: echo:Home XYZ First
Recv: //action:notification Home XYZ First
Recv: ok
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Send: M117 wait...
Recv: //action:notification wait...
Recv: ok
Send: G91
Recv: ok
Send: G28 X0 Y0 Z0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: X:154.00 Y:117.00 Z:12.73 E:0.00 Count X:12320 Y:9360 Z:5092
Recv: ok
Send: G90
Recv: ok
Send: G0 Z20
Recv: ok
Send: G29 D
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
[... 172 lines cut ...]
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: Bed X: 30.00 Y: 30.00 Z: 0.20
Recv: X:74.00 Y:37.00 Z:12.73 E:0.00 Count X:5920 Y:2960 Z:5092
Recv: ok
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Send: M117 0.20 >>> (adjust)
Recv: //action:notification 0.20 >>> (adjust)
Recv: ok
Send: G30 X30 Y30
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:23.00 /0.00 B:22.00 /0.00 @:0 B@:0
Recv: echo:busy: processing

j-be, 2022-03-31:
I see. That is actually what the printer reports. See the lines starting with `Recv: Bed X: 30.00 Y: 30.00`. The plugin just parses those lines and shows the information to the user.

I suspect there is an issue somewhere, either with the Z axis - see https://m.youtube.com/watch?v=2Z7mZVvPlc8 , or a wobbly X axis - see https://m.youtube.com/watch?v=4bFYH0X3qjk Alternatively it could be an issue with the BLTouch mount.

Also - as you mentioned you tightened everything: in my experience the wheels should only be as tight as absolutely needed. Overtightightening can cause all kind of issues.