j-be/AutoBim #33: Probe Deploy (M402) and Stow (M402)

adsaxton, 2022-04-11:
### Description

AutoBim does not deploy ABL before probing

### Steps to reproduce

When using probe such as the Allen Wrench type probe simply starting AutoBim will probe the mesh normally, deploying and stowing afterward. It will then proceed to attempt to probe the first corner without deploying the probe and fail.

### Expected outcome

For it to deploy the probe, execute the process and stow when finished.

### Actual outcome

As above in "Steps to reproduce"

### Environment (leave empty if unknown or not relevant)

* OctoPrint version: 1.7.3
* Python version:
* Marlin version: bugfix-2.0.x
* Probe (e.g. BL-Touch, EZABL, ...): Euclid Probe
* Printer (manufacturer/moodel, and please state any relevant modifications, like e.g. aftermarket mainboard): Ender 3 Pro
* Browser (especially useful on UI issues): Brave Browser

For UI issues a screenshot always helps.

## Feature

Add option to insert M401 before procedure and M402 after it's done.

**Please note by far the quickest way to get a new feature is to file a Pull Request.**

Just describe what you would like AutoBim as detailed as you can.

## Question

Make sure your question is not covered by [the Troubleshooting section](https://github.com/j-be/AutoBim#troubleshooting).

Else: just fire away. But keep in mind: the more detail, the easier it is to answer :smiley:.

j-be, 2022-04-11:
Hi, thanks for the report. I'm not quite sure what's going on here, but I think I could easily implement a "Do this before starting" and "Do that after completeing" with custom GCode. Maybe that would also solve #31.

Btw:

> AutoBim will probe the mesh normally

That should not happen. AutoBim should not probe the complete mesh. See https://github.com/j-be/AutoBim#what-is-supposed-to-happen for further detail. You might be having UBL enabled, but AutoBim failed to detect it. Be sure to double-check AutoBim's settings match your firmware configuration.

adsaxton, 2022-04-11:
> You might be having UBL enabled, but AutoBim failed to detect it.

I have BILINEAR enabled. Autobim has always done the initial mesh before starting. Even before with the BLTouch setup.

j-be, 2022-04-12:
That is very weird. The plugin does:

* `G29` (Home)
* Raise to Z=20
* Clear mesh (`G29 D` for UBL, `G29 J` for Bilinear)
* Go to first corner and do its thing

but at no point does it invoke a plain `G29`, i.e. "acquire mesh", as it makes no sense for the plugin to do so.

So what I think is happening here is that for some reason AutoBim detected UBL on first run and now goes for `G29 D` for "clear mesh". But `D` seems to be undefined for Bilinear (see https://marlinfw.org/docs/gcode/G029-abl-bilinear.html) so I guess the `D` is ignored and a plain `G29` is executed, which is not optimal.

So be sure to check AutoBim's settings - i.e. `Unified Bed Leveling` - and make sure it is off.

[image]

adsaxton, 2022-04-12:
-But D seems to be undefined for Bilinear

Actually, according to that link, **D** is defined as **Dry-Run mode. Just probe the grid but don’t update the bed leveling data**.

adsaxton, 2022-04-12:
So as it turns out the Unified Bed Leveling setting was the whole problem! It now Deploys and Stows. The problem is that it goes through this procedure with every probe! While watching the printer stow and deploy between each probe might be mildly entertaining when the bed is already tramed, it can be a real PIA between adjustments.

j-be, 2022-04-13:
I see, that is what I would have expected to begin with.

I just released `v0.2.3` which adds support for custom GCodes before and after each run. Now I hope Marlin is smart enough to keep state, i.e. it does not stow the probe if it was manually deployed before the first `G30`. Would you mind testing adding a `M401` before and a `M402` after?

adsaxton, 2022-04-13:
M401 and M402 are unnecessary. AutoBim effectively deploys prior to each probe and stows immediately afterward. This is a problem because it make it very slow and difficult to make the adjustments as it’s executing the next probe. With a little experimentation, I’ve discovered that if I probe a point using G30 with the additional parameter ‘E0’, it will probe without stowing afterward. It also doesn’t raise the probe afterward, so a ‘G0 Z15’ is necessary before executing the next probe command. I would try it in your code, but I have no experience with Java programming and aren’t sure how to test this.

j-be, 2022-04-14:
It is not so much what AutBim does - the plugin just tells Marlin `G30` and waits for the response.

Malin now goes "let me deploy, probe, and stow it again". I hoped, that maybe if we tell Marlin "deploy" (i.e. `M401`) and then tell it `G30` it would go "you told me to deploy, so I'm gonna probe and leave the probe deployed".

Could you please check if that is the case, i.e. does Marlin stow the probe after `G30` even if a `M401` was added to the "Before" GCode?

adsaxton, 2022-04-14:
I tried that. G30 will stow unless the option E0 is used.

j-be, 2022-04-14:
Damn, I still had hope...

Anyway, I'll see what I can do. I'll be on vacation next week, so not sure if I'll have any time for it. But I'll try to put together something that will work for probes like yours. I'll keep you updated 😄

closed 2022-04-28 by j-be

j-be, 2022-04-28:
I think I found a way to easily implement this. I pushed a preview of the feature to `next` branch.

@adsaxton Could you please test it?

To install, open OctoPrint's Settings and go to "Plugin Manager". There click on the "Get More" button on the top right, and under "... from URL" paste the following link `https://github.com/j-be/AutoBim/archive/next.zip` and click on "Install".

After that, you should have a new section in AutoBim's settings, like:

[image]

And don't forget to add the `M401` to "Custom GCode > Before" and `M402` to "Custom GCode > After".

j-be, 2022-04-29:
That is excellent news :smiley: 

I just released `v0.2.4` with this feature. OctoPrint should offer the update shortly.

Happy printing :smile:

commit 9ba00959 (2022-04-11): [Feature] Add support for custom before and after GCodes.

fixes #33
commit a534556d (2022-04-28): [Feature] Add support for custom probing sequences.

fixes #33