j-be/AutoBim #34: Hotend does not move after 2nd point.

RedBombX, 2022-05-22:
I have AutoBim on my Ender3v2 running Jyers and it runs just fine. I love it. 

However on my Ender3 Pro I have Marlin2.0 bugfix running and when I try to use AutoBim I will level the 1st and 2nd point just fine, but then the LCD display's that it's going to the next point and it the hotend doesn't move. I have to abort in Octoprint.

Wondering if this is a firmware issue?

closed 2022-05-22 by RedBombX

j-be, 2022-05-23:
@RedBombX Hi, sorry for the late answer. I figure you were hit by [this gotcha]?

If so, I'm really sorry for that. I know it is an issue, and I have an idea on how to solve it, but it turns out to be much more complicated than anticipated.

But it seems like I'll need a workaround for this, as it seems to cause a very bad first experience.

So in any case: thanks for the issue. If nothing else, I know I need to fix this asap :smile:

RedBombX, 2022-05-24:
Yes, I posted then I saw the FAQ, my bad. I still haven't gotten around to
trying to fix it.

I love AutoBim more than the Bed Leveling Wizard, but I'll have to use that
on my Pro if I cannot get AutoBim to work properly =(

Thanks for the reply!

j-be, 2022-05-24:
Happy you like it :smile: 

To fix this:
* open the OctoPrint settings
* select AutoBim in the left hand menu
* home the printer
* lower the X and/or Y values of the individual corners until a click on "Test" button makes the hotend move

Additionally, if you compile your own firmware, consider raising `X_MAX_POS` and `Y_MAX_POS` to 240. That work fine on my Ender 3 Pro and should give it a bit more space to move without colliding with anything.

RedBombX, 2022-05-24:
Oh that's excellent! I was trying to recompile and I couldn't figure out
what exactly needs changing and what is still safe that won't damage the
machine. I will be trying that today! Thanks!!

j-be, 2022-05-24:
You're welcome. Be careful though: after changing it slowly drive it to its new max positions 1mm at a time.

I'm using 245 for X and 240 for Y (see https://github.com/j-be/Ender3-Marlin-Config/blob/main/marlin/Configurations/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration.h#L1390-L1391), but I did move the Y endstop (that is why my ` Y_MIN_POS` is `-26`). So your results may vary.