j-be/AutoBim #36: Ender 3 S1 Pro Can't reach coordinates

Ajiirtj, 2022-11-22:
Hello!

I just installed TH3D's LCD screen and unified firmware on my s1 pro. In octoprint, when I send the command g0 x30 y30, the nozzle will move to that location but the probe is no longer on the bed. When I try to test that point in Autobim (after homing, first) it tells me that it cannot reach those very same coordinates.

When I try to manually change the coordinates, the max that I can reach in the autobim menu is still way off of the adjuster knob screws. I'm fairly new to modifying a printer this in-depth, and I have no idea how to fix the issue.

What are your thoughts?

Ajiirtj, 2022-11-22:
I should also mention, that when I use different coordinates in the autobim menu that the probe will reach, the coordinates do not match up with what is on my printer's screen. The coordinate's on my printers screen DO match up with what I see in octoprint, however. For instance, right now in autobim I have x50 y60, but on both my printer screen and octoprint it is showing x82, y101

j-be, 2022-11-22:
Hi, that all sounds like expected. In detail:

`G0` moves the _nozzle_ to the given coordinates, while AutoBim moves the _probe_ to the given coordinates. Assuming your probe to nozzle offsets are right this is fine. To check that:

* home
* move the _nozzle_ to X30 Y30 (`G0 X30 Y30`)
* mark the point right below the nozzle on the bed (e.g. using painters tape or whatever)
* in AutoBim enter X30 Y30 and hit test

The probe should now touch down exactly at the marked spot. If not, see https://marlinfw.org/docs/gcode/M851.html.

As for the "cannot reach the far end of the bed": there are 2 settings in Marlin: `BED_SIZE` and `MAX_POS` (example see [here](https://github.com/j-be/Ender3-Marlin-Config/blob/dfc55f2a08c0b9d7e8459096b1bba67253bdf9da/marlin/Configurations/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0/Configuration.h#L1687) ). Now, Creality sets both to same value, which makes no sense at all, as most printers can move beyond their bed, they just shouldn't print there. Unfortunately, this is a hard limit enforced by the firnware, there is no way around that, except for compiling your own firmware.

For completeness: there are other settings that could limit the area reachable by the probe, but in my experience the culprit is always `MAX_POS`.

TL;DR: The coordinates in AutoBim always refer to the probe's position, not the nozzle, and Creality makes shitty firmware.