j-be/AutoBim #37: requested feature probing failed

Diilandh, 2023-05-10:
Hello

first of all i really like this plugin.

my bltouch on one of my printers regularly fails at probing, normally i just restart my print and than it probes fine and the print starts
I use jyers firmware which probes 10x10 points when i make a mesh but when it fails probing it just restarts the failed probing point. 
Is it possible to add this feature to the plugin? because now when the probing fails i have to restart the whole procedure and start again at point 1.

thanks in advance

Sincerely

j-be, 2023-05-10:
Hi,

happy you like the plugin. I'd be happy to try and implement something like that. Could you please provide a log of what happens on such an error? You should be able to just copy it out of the respective tab in Octoprint.

Cheers

Diilandh, 2023-05-10:
Hello

I looked at my log and found this:

2023-05-10 11:21:51,170 - octoprint.plugins.autobim - INFO - Starting
2023-05-10 11:21:51,242 - octoprint.plugins.action_command_notification - INFO - Got a notification: wait...
2023-05-10 11:21:51,299 - octoprint.plugins.autobim - INFO - Updated G30 pattern to Marlin as the firmware name is Marlin bugfix-JyersUI v2.0.1 (Nov  5 2021 21:19:22)
2023-05-10 11:22:29,680 - octoprint.plugins.autobim - INFO - Updated G30 pattern to Marlin as the firmware name is Marlin bugfix-JyersUI v2.0.1 (Nov  5 2021 21:19:22)
2023-05-10 11:22:29,789 - octoprint.plugins.action_command_notification - INFO - Got a notification: 0.36 >>> (adjust)
2023-05-10 11:22:40,904 - octoprint.plugins.autobim - INFO - Updated G30 pattern to Marlin as the firmware name is Marlin bugfix-JyersUI v2.0.1 (Nov  5 2021 21:19:22)
2023-05-10 11:22:52,150 - octoprint.plugins.autobim - INFO - Updated G30 pattern to Marlin as the firmware name is Marlin bugfix-JyersUI v2.0.1 (Nov  5 2021 21:19:22)
2023-05-10 11:22:52,246 - octoprint.plugins.action_command_notification - INFO - Got a notification: 0.32 >>> (adjust)
2023-05-10 11:23:06,591 - octoprint.plugins.action_command_notification - INFO - Got a notification: Probing Failed
2023-05-10 11:23:06,593 - octoprint.util.comm - WARNING - Received an error from the printer's firmware: Probing Failed
| Last lines in terminal:
| Recv: ok
| Recv:  T:24.36 /0.00 B:50.00 /50.00 @:0 B@:47
| Send: M117 0.32 >>> (adjust)
| Recv: //action:notification 0.32 >>> (adjust)
| Recv: ok
| Send: G30 X30 Y30
| Recv:  T:24.43 /0.00 B:49.98 /50.00 @:0 B@:55
| Recv: echo:busy: processing
| Recv:  T:24.39 /0.00 B:50.01 /50.00 @:0 B@:44
| Recv: echo:busy: processing
| Recv:  T:24.39 /0.00 B:50.04 /50.00 @:0 B@:31
| Recv: echo:busy: processing
| Recv:  T:24.32 /0.00 B:50.01 /50.00 @:0 B@:38
| Recv: echo:busy: processing
| Recv:  T:24.38 /0.00 B:49.98 /50.00 @:0 B@:54
| Recv: echo:busy: processing
| Recv:  T:24.38 /0.00 B:50.02 /50.00 @:0 B@:34
| Recv: echo:busy: processing
| Recv: //action:notification Probing Failed
| Recv: Error:Probing Failed
2023-05-10 11:23:21,176 - octoprint.plugins.autobim - ERROR - Aborted by user
2023-05-10 11:23:21,191 - octoprint.plugins.autobim - INFO - 'None' from queue means user abort

when the probing fails it displays that the probing failed on the screen and than nothing happens so i abort and start again. 

sincerely

Diilandh, 2023-05-10:
hello

for example when i make a mesh it probes the points but when the probing fails my gantry goes up and it tries again at the same point. I have copied some of the data from the terminal.

Probing mesh point 5/100.
Probing mesh point 6/100.
Probing mesh point 7/100.
//action:notification Probing Failed
Error:Probing Failed
Probing mesh point 8/100.
Probing mesh point 9/100.
Probing mesh point 10/100.
Probing mesh point 11/100.
Probing mesh point 12/100.
Probing mesh point 13/100.
Probing mesh point 14/100.
Probing mesh point 15/100.
Probing mesh point 16/100.
Probing mesh point 17/100.
Probing mesh point 18/100.
Probing mesh point 19/100.
Probing mesh point 20/100.
//action:notification Probing Failed
Error:Probing Failed
Probing mesh point 21/100.
Probing mesh point 22/100.
Probing mesh point 23/100.
Probing mesh point 24/100.
Probing mesh point 25/100.
Probing mesh point 26/100.
//action:notification Probing Failed
Error:Probing Failed
Probing mesh point 27/100.
Probing mesh point 28/100.
Probing mesh point 29/100.
Probing mesh point 30/100.
Probing mesh point 31/100.
Probing mesh point 32/100.
Probing mesh point 33/100.
Probing mesh point 34/100.
Probing mesh point 35/100.
Probing mesh point 36/100.
Probing mesh point 37/100.

sincerely

j-be, 2023-05-10:
Perfect, that is exactly what I needed! I'll try to put something together. I'll come back to you once it is ready.

Cheers

j-be, 2023-05-12:
I just released version `v.0.3.0`, which should contain the feature: It will now retry every failed probing twice (so a maximum of 3 tries in total). OctoPrint should offer the update shortly.

Could you please check it? If it doesn't work as expected, feel free to reopen the branch.

Diilandh, 2023-05-12:
Wauw that is fast :D. 

I just tried the new version and it works perfectly. Thank you so much. 

its thanks to people like yourself that this hobby is so worthwhile.

sincerely

j-be, 2023-05-12:
My pleasure, happy printing :smile: