j-be/AutoBim #39: Stuck on Wait..... Point X Y seems to be unreachable

SoaringEagle93, 2023-09-26:
## Bug report

Please do not report the "Works fine on the first (or first and second corner), display says 'ok. moving to next' but
nothing happens" bug. This is covered in detail in the "Troubleshooting" section of the README. I know it is an issue
for many, and it is on the roadmap, but unfortunately it isn't an easy fix.

### Description

When Autobim is started it will home and then move to the first corner, probe, then move and say "wait." Also when I go into the settings and probe each test point it successfully probes, however it tells me "Point X30 Y30 seems to be unreachable!"

### Steps to reproduce

Press the AutoBim button.
Also going into the settings and testing each point.

### Expected outcome

I would expect it to probe then tell me how much to adjust before probing again.

### Actual outcome

Sit on the screen that says wait.

### Environment (leave empty if unknown or not relevant)

* OctoPrint version: 1.9.2
* Python version:
* Marlin version: 2.1.x
* Probe (e.g. BL-Touch, EZABL, ...): BL-Touch
* Printer (manufacturer/moodel, and please state any relevant modifications, like e.g. aftermarket mainboard): Ender 3 Pro with SKR mini E3 v3.0
* Browser (especially useful on UI issues):

For UI issues a screenshot always helps.

j-be, 2023-09-27:
Hi,

> I would expect it to probe then tell me how much to adjust before probing again.

Unfortunately, it can't do that. This depends on settings in the firmware, that can afaik. not be queried, like `X_MIN_POS`. But if you know those you should be able to easily compute them. Also, be sure that your probe-to-nozzle offset is correct. See https://github.com/j-be/AutoBim#works-fine-on-the-first-or-first-and-second-corner-display-says-ok-moving-to-next-but-nothing-happens for details.

j-be, 2023-09-27:
Oh, and could you please provide the terminal output while probing?

I.e. start a probe (e.g. using the "Test" button in  "Settings") and afterwards go to OctoPrint's "Terminal" tab and copy the last couple of lines and post them here.

Please make sure to at least include the output starting from the `G30` command.

SoaringEagle93, 2023-09-28:
My X_MIN_POS and Y_MIN_POS are 0,0 in the firmware and when homing XY the nozzle is over the front left corner as it should be.  

When I run the test after Homing this is what I get in the terminal:

[...]
Send: G30 X30 Y30
[...]
Recv: Bed X:30.0 Y:30.0 Z:0.09
[...]
Recv: ok

j-be, 2023-09-28:
Wow, that's interesting: it seems like your firmware - different from anything I have seen yet - does not put spaces between the coordinates and the values in the G30 respomse. I.e., I have always seen:

`Bed X: 30.0 Y: 30.0 Z: 0.09`

but yours is

`Bed X:30.0 Y:30.0 Z:0.09`

Could you please try to put the following in Settings -> Custom G30 response pattern:

`^Bed X:-?\d+\.\d+ Y:-?\d+\.\d+ Z:(-?\d+\.\d+)$`

and see if it works then? If so, it should be an easy fix.

Also, what firmware are you running? Like exact version in case of self-compiled or place where you downloaded it from?

SoaringEagle93, 2023-09-28:
I added the custom G30 response pattern and got the same result. It probes the spot but then says it is unreachable. 
My firmware is self compiled and is version bugfix-2.1.x (Jul 30 2023). I used the Creality/Ender-3 Pro/BigTreeTech SKR Mini E3 3.0 example to build it.

j-be, 2023-09-28:
Weird... I think I have some time tomorrow to dig deeper. I may need to increase logging verbosity to see what is going on here.

I'll keep you up to date if something comes up. To be honest: I'm at the moment not quite sure what is going on here.

j-be, 2023-09-29:
Alright, I think I fixed it. I just releases version `0.4.1` - OctoPrint should offer the update shortly (I think it caches release data for 24 hours).

Please countercheck if it is now working at your side, too. Please clear "Custom G30 response pattern" when doing so - should now work out of the box.

SoaringEagle93, 2023-09-30:
That seem to do it. Now it tells me each point is available and when I click the "Start AutoBim" everything acts as it should. Thanks for looking into and fixing this so quickly!

j-be, 2023-09-30:
My pleasure. Thanks for reaching out. This would have caused issues to a lot of people with the next Marlin release.

closed 2023-09-30 by j-be