j-be/AutoBim #41: Ender 3 V2 - Failed to Enable Bed Leveling

macharborguy, 2025-01-25:
First time using AutoBIM on my Ender 3 V2.  Im using a status plugin for OctoPrint that will show the status messages for making adjustments, and after the second adjustment sequence (first = lower left, second = upper right), it doesnt progress to the next corner, I get a popup error that says that the printer failed to enable bed leveling, and the communication code shows as such...

```Recv:  T:22.15 /0.00 B:21.76 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.23 /0.00 B:21.56 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.15 /0.00 B:21.60 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:22.11 /0.00 B:21.68 /0.00 @:0 B@:0
Recv:  T:22.19 /0.00 B:21.48 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.19 /0.00 B:21.52 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv:  T:22.03 /0.00 B:21.64 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.19 /0.00 B:21.48 /0.00 @:0 B@:0
Recv: X:157.00 Y:122.00 Z:10.00 E:0.00 Count X:12560 Y:9760 Z:4000
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Operational" to "Error"```

I then have to power cycle the printer to reconnect it to OctoPrint.

OctoPrint Version : 1.10.3
AutoBIM Version : 0.4.1
Ender 3 v2 Firmware Version : v1.0.7 (official firmware)

All AutoBIM settings are defaults.

j-be, 2025-01-25:
Heyhey,

first guess: third corner is not reachable. This is unfortunately quite tricky to detect - I did still not come up with a logic that detects this reliably.

Please got to the AutoBim Settings and adjust the coordinates for the third corner until "Test" button next to it can successfully probe it:

[image]

Please do not change any other setting for the time being, even if they differ from what you see in the screenshot.

macharborguy, 2025-01-25:
thanks!  I was quick to post, and didnt think to look in the closed issues.  Found this one (https://github.com/j-be/AutoBim/issues/34#issuecomment-1135866375)  and just finished configuring via the corner testing and got a successful leveling process.

The values that worked for me was to set the X for both the third and fourth corner to 195 instead of the default 200.  (196 - 199 also didnt work, but 195 hit the sweet spot)

closed 2025-01-25 by macharborguy

j-be, 2025-01-25:
Awesome, good to hear 😄 

Happy printing!